raspberrypi/openocd #104: Can't connect to rp2350 riscv core via jlink

simonjwright, 2024-08-23:
macOS Sonoma 14.6.1 (M1)
Command Line Tools 15.3
Homebrew
This repo sdk-2.0.0
libjaylink 0.3.1
gcc 14.1.0 (aarch64)
jlink-edu v9.3
pico 2

`./configure --prefix=$HOME/local --enable-jlink --enable-internal-jlink`

Running with this script:
```
OPENOCD_LOC=~/local
$OPENOCD_LOC/bin/openocd                        \
    -f interface/jlink.cfg                      \
    -c "adapter speed 5000"                     \
    -c "set USE_CORE 0"                         \
    -f target/rp2350-riscv.cfg
```
I get
```
adapter speed: 5000 kHz
0
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : J-Link V9 compiled May  7 2021 16:26:12
Info : Hardware version: 9.30
Info : VTarget = 3.240 V
Info : clock speed 5000 kHz
Info : SWD DPIDR 0x4c013477
Info : [rp2350.dap.core0] datacount=1 progbufsize=2
Info : [rp2350.dap.core0] unavailable.
Error: [rp2350.dap.core0] Examination failed
Warn : target rp2350.dap.core0 examination failed
Info : starting gdb server for rp2350.dap.core0 on 3333
Info : Listening on port 3333 for gdb connections
^Cshutdown command invoked
Error: [rp2350.dap.core0] Unsupported DTM version: -1
Error: [rp2350.dap.core0] Could not identify target type.
Error: BUG: refcount AP#0 still 1 at exit
```
(including the last 3 lines for info).

If I use `target/rp2350.cfg`, I get
```
adapter speed: 5000 kHz
0
Info : Hardware thread awareness created
cortex_m reset_config sysresetreq
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : J-Link V9 compiled May  7 2021 16:26:12
Info : Hardware version: 9.30
Info : VTarget = 3.245 V
Info : clock speed 5000 kHz
Info : SWD DPIDR 0x4c013477
Info : [rp2350.dap.core0] Cortex-M33 r1p0 processor detected
Info : [rp2350.dap.core0] target has 8 breakpoints, 4 watchpoints
Info : [rp2350.dap.core0] Examination succeed
Info : starting gdb server for rp2350.dap.core0 on 3333
Info : Listening on port 3333 for gdb connections
^Cshutdown command invoked
```

mookiedog, 2024-09-16:
I am seeing this same issue using an official Pi CMSIS-DAP Debug Probe.  I diff'd the rp2350.cfg (arm) and rp2350-riscv.cfg files and something looks maybe a bit weird:
[image]

To my totally untrained eye, the diff above looks like the risc-v version got the core0 'if' modified for risc-v, then cut&pasted to create the core1 'if' because both the 'if' statements are commented as '# core 0". But the part that seems iffy is that the ap-num parameter is identical for both cores on riscv, while the arm version shows 2 different values. That doesn't seem right, but I am clutching at straws at this point.

mookiedog, 2024-09-16:
I think my previous comments were mostly inaccurate. But I have discovered this: if I burn and flash a UF2 file that executes a risc-v program, then while that program is running, it is possible to get past the examination phase that was failing originally:

openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -f /home/robin/.vscode-server/extensions/marus25.cortex-debug-1.12.1/support/openocd-helpers.tcl -f interface/cmsis-dap.cfg -f target/rp2350-riscv.cfg -c "adapter speed 5000"
Open On-Chip Debugger 0.12.0+dev-gebec9504d (2024-09-14-17:53)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDLiveWatchSetup
Info : Hardware thread awareness created
Info : Hardware thread awareness created
adapter speed: 5000 kHz
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E6616408436A7E22
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 5000 kHz
Info : SWD DPIDR 0x4c013477
Info : [rp2350.dap.core0] datacount=1 progbufsize=2
Info : [rp2350.dap.core0] Disabling abstract command reads from CSRs.
Info : [rp2350.dap.core0] Disabling abstract command writes to CSRs.
Info : [rp2350.dap.core0] Core 0 could not be made part of halt group 1.
Info : [rp2350.dap.core0] Examined RISC-V core
Info : [rp2350.dap.core0]  XLEN=32, misa=0x40901105
Info : [rp2350.dap.core0] Examination succeed
Info : [rp2350.dap.core1] datacount=1 progbufsize=2
Info : [rp2350.dap.core1] Disabling abstract command reads from CSRs.
Info : [rp2350.dap.core1] Disabling abstract command writes to CSRs.
Info : [rp2350.dap.core1] Core 1 could not be made part of halt group 1.
Info : [rp2350.dap.core1] Examined RISC-V core
Info : [rp2350.dap.core1]  XLEN=32, misa=0x40901105
Info : [rp2350.dap.core1] Examination succeed
Info : starting gdb server for rp2350.dap.core0 on 50000
Info : Listening on port 50000 for gdb connections

lurch, 2024-09-18:
> both the 'if' statements are commented as '# core 0".

See #105 :wink: 

> But the part that seems iffy is that the ap-num parameter is identical for both cores on riscv, while the arm version shows 2 different values.

See sections 3.5.2. and 3.5.3. in https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf

simonjwright, 2024-09-18:
My interest is in building a [bare-metal Ada runtime](https://github.com/simonjwright/FreeRTOS-Ada/tree/dev/rp2350), based on vanilla FreeRTOS, on a single risc-v core. I did manage to work, in a clumsy and uncomfortable way, using a no_flash.uf2 risc-v executable that some kind soul had uploaded, but of course every time I had to reset (I hadn't at that point spotted `monitor reset init`) I had to go through the dance of reloading the UF2 and then starting openocd.

I eventually found CRIT1_BOOT_ARCH in the data sheet (3.9), which makes the default boot architecture risc-v.

It took me a while to find the picotool  setting (`picotool opt set crit1.boot_arch`). Much happier now.

By the way, with this change the chip runs at 50 MHz, not 150!! Is this a known issue? Should I report it? (how?)

lurch, 2024-09-18:
> By the way, with this change the chip runs at 50 MHz, not 150!! Is this a known issue? Should I report it? (how?)

Best place would be https://github.com/raspberrypi/pico-feedback/issues