raspberrypi/openocd #107: Mac M1 Build fails: Invalid casts from uint8_t * to uint32_t *

jondurrant, 2024-08-25:
Building on Mac M1 with clang-1500.3.9.4 I get the following error:

src/target/riscv/batch.c:194:24: error: cast from 'uint8_t *' (aka 'unsigned char *') to 'uint32_t *' (aka 'unsigned int *') increases required alignment from 1 to 4 [-Werror,-Wcast-align]
                        uint32_t *data_in = (uint32_t *)batch->data_in;

As batch->data_in is uint8_t * this doesn't make a lot of sense. Reading the code I can't work out what is meant to be trying to do.

AndreyLalaev, 2024-08-30:
If this issue is still actual, apply this patch & rebuild: [0001-riscv-batch.c-fix-compilation-error-on-macos.patch](https://github.com/user-attachments/files/16818715/0001-riscv-batch.c-fix-compilation-error-on-macos.patch)

jalmeroth, 2024-09-06:
I had the same issue on Linux and went through all files where `calloc(sizeof…` appeared and switched the order of the parameters. Here's the resulting patch that made it work for me [calloc-sizeof.patch.txt](https://github.com/user-attachments/files/16912858/calloc-sizeof.patch.txt).

Hope this helps

Uthedris, 2025-05-23:
This error still exists as of the date of this comment.   23 May 2025

The patch mentioned above fixes the error.  Thank You!

BTW, to apply patch after download, from the openocd directory use:
```
patch -p 1 -i ~/Downloads/0001-riscv-batch.c-fix-compilation-error-on-macos.patch
```