raspberrypi/openocd #110: RP2350 - Upload to and run from SRAM not working

ChristianIvarsson, 2024-09-07:
Hi!

It has previously been possible to upload to and run from SRAM on the older rp2040 but it seems no longer possible on the new sibling.

Is there any planned support in the pipeline or would it be better if init were performed by the script itself?

Expected working configuration:
```
<CMakeLists.txt>
add_compile_definitions(PICO_NO_FLASH=1)
.. ..
pico_set_binary_type(${PROJECT_NAME} no_flash)

<ramload.cfg>
source [find interface/cmsis-dap.cfg]
adapter speed 4000
source [find target/rp2350.cfg]

transport select swd
init
reset halt

load_image image.elf
resume 0x20000000
shutdown
```

~I'm really sorry if this is a PEBCAK

ChristianIvarsson, 2024-09-17:
Did a test on the built elf just to verify and it seems the sdk does what it should
arm-none-eabi-size -A -x yields 
```
section                  size         addr
.text                  0x5338   0x20000000
.rodata                 0x410   0x20005338
.binary_info             0x1c   0x20005748
.data                   0x218   0x20005770
.tdata                    0x0   0x20005988
.uninitialized_data       0x0   0x20005988
.tbss                     0x0   0x20005988
.bss                    0x54c   0x20005988
.heap                 0x3a12c   0x20005ed4
.. ..
```

Also tried openocd on a pi through its gpio interface and it was the same result there.

lurch, 2024-09-18:
This _might_ be related to https://github.com/raspberrypi/pico-feedback/issues/153 ? :man_shrugging:

ChristianIvarsson, 2024-09-25:
> This _might_ be related to [raspberrypi/pico-feedback#153](https://github.com/raspberrypi/pico-feedback/issues/153) ? 🤷‍♂️
( I must sort out my mail situation. I'm really sorry for such a late reply )

It's at the very least VERY closely related. Think they only differ in that I want to start and instantly let go of the target instead of attaching a debugger. Everything up to that point ought to have the same requirements.

I'll see if I can create some test cases to verify if it's just me doing something weird (most likely) or if it's currently not possible to debug and/or run from sram.