raspberrypi/openocd #119: Unkown error at file "embedded:startup.tcl", line 28

arenick001, 2024-11-10:
following is my config:
```
tcl_port disabled
telnet_port disabled
adapter driver ftdi

adapter speed 15000

adapter serial $_SERIAL

ftdi vid_pid 0x0403 0x6010
transport select jtag

ftdi channel 0

ftdi layout_init 0x0088 0x008b
reset_config none
set _CHIPNAME xc7

jtag newtap $_CHIPNAME tap -irlen 6 -ignore-version \
	-expected-id 0x03622093 \
	-expected-id 0x03620093 \
	-expected-id 0x037C4093 \
	-expected-id 0x0362F093 \
	-expected-id 0x037C8093 \
	-expected-id 0x037C7093 \
	-expected-id 0x037C3093 \
	-expected-id 0x0362E093 \
	-expected-id 0x037C2093 \
	-expected-id 0x0362D093 \
	-expected-id 0x0362C093 \
	-expected-id 0x03632093 \
	-expected-id 0x03631093 \
	-expected-id 0x03636093 \
	-expected-id 0x03647093 \
	-expected-id 0x0364C093 \
	-expected-id 0x03651093 \
	-expected-id 0x03747093 \
	-expected-id 0x03656093 \
	-expected-id 0x03752093 \
	-expected-id 0x03751093 \
	-expected-id 0x03671093 \
	-expected-id 0x03667093 \
	-expected-id 0x03682093 \
	-expected-id 0x03687093 \
	-expected-id 0x03692093 \
	-expected-id 0x03691093 \
	-expected-id 0x03696093

# usage: pld create pld_name driver_name -chain-position tap_name [driver_options]
pld create $_CHIPNAME.pld virtex2 -chain-position $_CHIPNAME.tap -no_jstart
virtex2 set_user_codes $_CHIPNAME.pld 0x02 0x03 0x22 0x23

init

virtex2 program $_CHIPNAME.pld
pld load $_CHIPNAME.pld $_BIT_FILE_PATH
exit
```
the error accurred at line `pld create $_CHIPNAME.pld virtex2 -chain-position $_CHIPNAME.tap -no_jstart`

I am a newbie to FPGA and hope someone can help me solve this problem. PLS

lurch, 2024-11-11:
> I am a newbie to FPGA...

This is Raspberry Pi's fork of OpenOCD, which is intended to be used with SWD and our [Debug Probe](https://www.raspberrypi.com/products/debug-probe/) in order to flash our [microcontoller](https://www.raspberrypi.com/products/raspberry-pi-pico/) [boards](https://www.raspberrypi.com/products/raspberry-pi-pico-2/).

It sounds like you're probably looking for the upstream [OpenOCD project](https://openocd.org/).

closed 2024-11-11 by lurch

arenick001, 2024-11-11:
I don't know if it's related... I'm currently trying to use openocd in a raspberry pi to operate an FPGA connected to it

lurch, 2024-11-11:
Okay, to clarify: the _sole_ purpose of the repo is to provide support for using SWD to program and debug the Raspberry Pi Pico or Raspberry Pi Pico 2 **microcontroller** boards (or any 3rd-party microcontroller boards that use our RP2040 or RP2350 chips).
Simply running OpenOCD on a Raspberry Pi ["flagship" board](https://www.raspberrypi.com/news/raspberry-pi-product-series-explained/) definitely falls under the remit of the upstream OpenOCD project (especially as you're using JTAG) :slightly_smiling_face: 

So even if there **is** "something different about the Raspberry Pi" then it doesn't fall under the umbrella of this repository.