raspberrypi/openocd #120: Warning: sysresetreq not set on rp2350.dap.core1

cibomahto, 2024-11-21:
If I build OpenOCD from the sdk-2.0.0 branch, and use it with a Raspberry Pi Debug Probe to flash a binary to an RP2350, I get the following warnings:

```
/usr/local/bin/openocd -f interface/cmsis-dap.cfg -f target/rp2350.cfg -c "adapter speed 5000" -c "program src/main.elf verify ; reset ; exit"

Open On-Chip Debugger 0.12.0+dev-gebec950 (2024-11-16-18:36)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : Hardware thread awareness created
cortex_m reset_config sysresetreq
adapter speed: 5000 kHz
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E6632891E3428C30
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 5000 kHz
Info : SWD DPIDR 0x4c013477
Info : [rp2350.dap.core0] Cortex-M33 r1p0 processor detected
Info : [rp2350.dap.core0] target has 8 breakpoints, 4 watchpoints
Info : [rp2350.dap.core0] Examination succeed
Info : [rp2350.dap.core1] Cortex-M33 r1p0 processor detected
Info : [rp2350.dap.core1] target has 8 breakpoints, 4 watchpoints
Info : [rp2350.dap.core1] Examination succeed
Info : starting gdb server for rp2350.dap.core0 on 3333
Info : Listening on port 3333 for gdb connections
Warn : [rp2350.dap.core1] VECTRESET is not supported on this Cortex-M core, using SYSRESETREQ instead.
Warn : [rp2350.dap.core1] Set 'cortex_m reset_config sysresetreq'.
[rp2350.dap.core0] halted due to breakpoint, current mode: Thread
xPSR: 0xf9000000 pc: 0x00000088 msp: 0xf0000000
[rp2350.dap.core1] halted due to debug-request, current mode: Thread
xPSR: 0xf9000000 pc: 0x00000088 msp: 0xf0000000
** Programming Started **
Info : RP2040 Flash Probe: 33554432 bytes @0x10000000, in 8192 sectors

Info : Padding image section 1 at 0x10000ce4 with 28 bytes (bank write end alignment)
Warn : Adding extra erase range, 0x10000d00 .. 0x10000fff
** Programming Finished **
** Verify Started **
** Verified OK **
Warn : [rp2350.dap.core1] VECTRESET is not supported on this Cortex-M core, using SYSRESETREQ instead.
Warn : [rp2350.dap.core1] Set 'cortex_m reset_config sysresetreq'.
```

In the [rp2350 target](https://github.com/raspberrypi/openocd/blob/sdk-2.0.0/tcl/target/rp2350.cfg#L79), it appears that this is meant to be set, but it seems to be applied to the first core only. If I configure sysresetreq for the second core specifically:

```
-c "rp2350.dap.core1 cortex_m reset_config sysresetreq"
```

then the warning goes away. I'm not familiar enough with the configuration system to know if the second core should be configured explicitly in the rp2350 config file, or if there is a bug somewhere else.

PaulskPt, 2025-02-24:
I have the same problem. I also don't know how to set this specific item in whatever configuration file.

cibomahto, 2025-02-27:
@PaulskPt I've just been specifying it on the command line to launch openocd. Here's a somewhat over-complicated example:

https://github.com/Blinkinlabs/rp2350_tinygo_dev/blob/42842292f174d4a9e6339a2a9435e67a22e3ff93/docker/Makefile#L46

PaulskPt, 2025-03-03:
@cibomahto. Thank you for your message. What I need is to know where I can add the command line: -c "rp2350.dap.core1 cortex_m reset_config sysresetreq" inside the Arduino IDE v2.3.4. I searched in places like: 
<drive>:\Users\<user>\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\4.0.1-8ec9d6f, however I didn't find where I could add a command like mentioned.

cibomahto, 2025-03-03:
> [@cibomahto](https://github.com/cibomahto). Thank you for your message. What I need is to know where I can add the command line: -c "rp2350.dap.core1 cortex_m reset_config sysresetreq" inside the [...]

I'm not using Arduino so I don't know specifically, but try searching those directories for another part of the command, such as "interface/cmsis-dap.cfg". That should help narrow down where the configuration is.

PaulskPt, 2025-03-03:
Thank you. I go to look in there.

lurch, 2025-03-04:
> where I can add the command line: -c "rp2350.dap.core1 cortex_m reset_config sysresetreq" inside the Arduino IDE v2.3.4

Perhaps @earlephilhower can help?

earlephilhower, 2025-03-04:
Not really sure that the Arduino core (which already has its own OpenOCD core built from the new SDK and RPI's OpenOCD fork) would ever call your own custom built OpenOCD.  I suppose you could hack file places, overwriting the Arduino package manager installed copy, but that's not a good long term plan.  FWIW, the distributed OpenOCD app also does uploads (and believe me, plenty of debugging too 😆 ) from the IDE w/o any issue that I've heard about.

In any case you can look in the installed core's `lib/rp2350(or rp2350-riscv)/picoprobe_cmsis_dap.tcl` which is what's used to start OpenOCD for debugging or the ` tools.picoprobe_cmsis_dap.upload.pattern=...` command line in `platform.txt`

PaulskPt, 2025-03-06:
@earlephilhower, thank you for your message. In this moment I am using MS Windows 11 Pro. board to test: iLabs cPico (RP2350 with a 3-wire debug connector), an iLabs iProbe (clone of the PicoProbe). Using Arduino IDE v2.3.4. As far as I know, I am not using an "own custom built OpenOCD". In the Arduino IDE v2.3.4 settings, your library https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json. In the sketch menu > Optimize for debugging and in the tools menu > Upload method > Picoprobe/Debuggingprobe (CMSIS-DAP). I use a Picoprobe clone from iLabs (product name: iProbe). Debugging is functioning. The only thing is the error mentioned above regarding core1 reset_config sysresetreq.

earlephilhower, 2025-03-06:
Ah, I think I see.  Two different users using two different builds of OpenOCD (one from source, one from the arduino-pico core).  But both do generate the same warning (which makes sense since I don't patch the source, just take it and build for the ~9 architectures).

In any case the warning message seems to be just a warning and not a real problem.  Uploads work fine and multicore operations also seem good in all my testing.  You can edit those files I suggested or ignore the warning, it seems benign.

That said the TCL for the RP2350 here probably needs a patch to remove the warning as @cibomahto suggested.  #127 implements his suggestion and does clear things up in testing here.

PaulskPt, 2025-03-07:
I added the line as in  [#127](https://github.com/raspberrypi/openocd/pull/127) in rp2350.cfg. Now the warning is gone. Tnx!

light655, 2025-03-11:
> Ah, I think I see. Two different users using two different builds of OpenOCD (one from source, one from the arduino-pico core). But both do generate the same warning (which makes sense since I don't [...]

It's seems that core 1 is not properly reset in my testing.

I ran stdio on core 1 on the RP2350. After uploading the binaries using OpenOCD, the input and output are not produced (both USB and UART are checked). I need to reset the target RP2350 by shorting RUN to GND to have input/output working normally.

Here is the code used in my testing.
```
#include <stdio.h>
#include "pico/stdlib.h"
#include "pico/multicore.h"

void core1_main(void) {
    for(int i = 0; i < 10; i++) {
        printf("Hello, it's me.\n");
        sleep_ms(1000);
    }

    while(true) {
        int tmp = getchar_timeout_us(0);
        if(tmp == PICO_ERROR_TIMEOUT) {
            sleep_ms(5);
        } else {
            putchar(tmp);
        }
    }

    return;
}

int main(void) {
    stdio_init_all();

    multicore_launch_core1(core1_main);

    while(true) {
        tight_loop_contents();
    }

    return 0;
}
```

The OpenOCD version is 0.12.0+dev-gcf9c0b41c and the command used for uploading the binaries is:
`openocd -f interface/cmsis-dap.cfg -c "adapter speed 5000" -f /target/rp2350.cfg -c "program $(dirname $0)/build/picoprobe_stdio_test.elf verify reset exit"`

earlephilhower, 2025-03-11:
I don't think anyone here is reporting that, @light655 .  My own multicore I/O tests (using the Arduino core and rp2350) run just fine with or without the extra line to remove the warning (AIUI OpenOCD is doing the exact same thing in both cases, anyway, just removing a warning).  You should double check your setup (especially serial ports -> when the chip is reset asynchronously like w/OpenOCD the host USB stack may take some time to reconnect after it runs and may give it a different devnode (COM*: or /dev/tty*).  If that's still the case then the [forum](https://forum.arduino.cc/) or the [SDK repo](https://github.com/raspberrypi/pico-sdk) would be the right spot for it...

light655, 2025-03-12:
I've also see the same [rp2350.dap.core1] warnings so I thought it belongs here.
I've checked that the port is still the same after resetting. Also, the issue only occurs when the target is RP2350, RP2040 doesn't have any issue.
Thank you for the information, I'll report it on the forum or the SDK repo.

cibomahto, 2026-04-01:
Great! Just tried it, I can confirm that the warning is gone on the sdk-2.2.0 branch.