raspberrypi/openocd #125: Openocd (default branch (sdk-2.0.0)) Won't build from source on Ucrt64/MSYS2 or MinGW64/MSYS2 but will with WSL (not cross compile) due to uint in src/flash/nor/rp2040.c

SagaraBattousai, 2025-02-21:
When trying to build Openocd following both the instructions at getting-started-with-pico.pdf and README.Windows I am unable to build the default branch (sdk-2.0.0) with either UCRT64/MSYS2 or MinGW64/MSYS2 as both result in error due to the use of ```uint``` in lines 192, 236, 414 and 491 of src/flash/nor/rp2040.c. Interestingly though, there are no problems compiling with WSL2 (although that is for a direct compilation, no cross-compiling).

Fortunately, the  rp2040-0.12-rc2 branch compiles perfectly on UCRT64 (and from a quick skim of the source code has removed all uses of plain ```uint```, but I was not quite able/willing to try and merge the two branches locally).

I apologise if this is due to a mistake on my behalf, it has been a long long time since I've built anyone else's software, I probably should just have learnt to cross-compile :)

SagaraBattousai, 2025-02-21:
Sorry, possible self answer. Just saw this line of code in pico-sdk-tools ```sed -i -e 's/uint /unsigned int /g' ./src/flash/nor/rp2040.c``` which might sort it. (I should probably just consign myself to use the vscode extension)

lurch, 2025-02-24:
You can also download precompiled binaries from https://github.com/raspberrypi/pico-sdk-tools/releases , if that meets your needs?

earlephilhower, 2025-03-13:
FWIW, can confirm MSYS2 builds have this problem for me, too, using GH CI.  I got around it by adding the definition to the generated `config.h` instead of changing the actual source files so the `git describe` would be clean:
https://github.com/earlephilhower/pico-quick-toolchain/blob/71f729a094edf4eca54a49ef40e4465f417d06d7/.github/workflows/build-windows.yml#L95-L109

lurch, 2025-03-13:
Ahhh, see also #108