raspberrypi/openocd #128: Build on mac fails

CrabbyPete, 2025-03-20:
When I run make on the Mac I get the following error:

/Library/Developer/CommandLineTools/usr/bin/make  all-recursive
Making all in jimtcl
	LINK	jimsh
ld: archive member '/' not a mach-o file in '/Users/xxx/.pico-sdk/openocd/jimtcl/libjim.a'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [jimsh] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2

lurch, 2025-03-20:
> ld: archive member '/' not a mach-o file

That's a weird error-message. Is there a space character (or any other unusual characters) in the `/xxx/` part of your path? Does the `.pico-sdk` mean that you're using https://github.com/raspberrypi/pico-vscode ?

CrabbyPete, 2025-03-21:
Yes the path had my full name where the xxx, I just x'd it out.  I am using  https://github.com/raspberrypi/pico-vscode  and its in my .pico-sdk path.

CrabbyPete, 2025-03-21:
gcc --version
Apple clang version 15.0.0 (clang-1500.1.0.2.5)
Target: x86_64-apple-darwin22.6.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin

lurch, 2025-03-24:
If you're using the pico-vscode extension, then I believe openocd _should_ be installed automatically, without you needing to build it from source? ping @will-v-pi 
Alternatively, you can also download prebuilt binaries from https://github.com/raspberrypi/pico-sdk-tools/releases

lurch, 2025-03-24:
Ahhh, but if you **do** want to build from source, perhaps one of these will help? 🤷 
https://stackoverflow.com/questions/77444892/xcode-15-0-1-macos-sonoma-clang-archive-or-linking-issue
https://github.com/pyenv/pyenv/issues/2862

CrabbyPete, 2025-03-24:
Thank, The one automatically installed does not have https://github.com/raspberrypi/openocd/blob/sdk-2.0.0/tcl/target/rp2350.cfg, and I get an error when I try to debug the Pico 2. I'll give this a shot

CrabbyPete, 2025-03-25:
I tried changing ar from /usr/local/bin to the one in /usr/bin. Same result