raspberrypi/openocd #129: Cannot connect to RP2350 in non-secure debug mode

PetteriAimonen, 2025-03-21:
When trying to debug RP2350 in non-secure debug mode using picoprobe, I get this error:

```
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E660443043813128
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 5000 kHz
Info : SWD DPIDR 0x4c013477
Info : SWD DPIDR 0x4c013477
Error: Failed to read memory at 0xe000ed00
Error: [rp2350.dap.core0] Examination failed
Warn : target rp2350.dap.core0 examination failed
Info : SWD DPIDR 0x4c013477
Error: Failed to read memory at 0xe000ed00
Error: [rp2350.dap.core1] Examination failed
Warn : target rp2350.dap.core1 examination failed
Info : starting gdb server for rp2350.dap.core0 on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Error: Target not examined yet
Error executing event gdb-attach on target rp2350.dap.core0:
```

I think this is due to wrong value in `CSW` register. Specifically, with `-d4` logging I see this:

```
Debug: 297 72 cmsis_dap.c:825 cmsis_dap_swd_write_from_queue(): AP write reg 0 a2000012
Debug: 298 72 cmsis_dap.c:825 cmsis_dap_swd_write_from_queue(): AP write reg 4 e000ed00
```

While if I try with openocd tcl interface without using the memap functions, I see CSW register defaults to `0x43000000` and I am able to access the memory address:

```
> rp2350.dap apreg 0x2000 0xd00
0x43000042    # CSW value
> rp2350.dap apreg 0x2000 0xd04 0xe000ed00  # Set transfer address register
> rp2350.dap apreg 0x2000 0xd0c   # Read data register
0x411fd210  # Correct CPUID value
```

I have tried adding the `rp2350.dap apcsw 0x43000000` command to `rp2350.cfg`, but it does not seem to apply.
The logs still show openocd overwriting `CSW` with `0xa2000012` which is the `CSW_AHB_DEFAULT` value.

I then tried hacking it into the source code:

```
diff --git a/src/target/arm_adi_v5.h b/src/target/arm_adi_v5.h
index 92c3dbc3a..d36bead3a 100644
--- a/src/target/arm_adi_v5.h
+++ b/src/target/arm_adi_v5.h
@@ -190,7 +190,8 @@
  * AHB3: SBO, UNPREDICTABLE if zero */
 #define CSW_AHB_SPROT           (1UL << 30)
 /* AHB: initial value of csw_default */
-#define CSW_AHB_DEFAULT         (CSW_AHB_HPROT1 | CSW_AHB_MASTER_DEBUG | CSW_DBGSWENABLE)
+//#define CSW_AHB_DEFAULT         (CSW_AHB_HPROT1 | CSW_AHB_MASTER_DEBUG | CSW_DBGSWENABLE)
+#define CSW_AHB_DEFAULT 0x43000000
 
 /* AXI: Privileged */
 #define CSW_AXI_ARPROT0_PRIV    (1UL << 28)
```

and that successfully examines the target and GDB is able to connect and debug.

**But how to set this value without recompiling openocd?**

PetteriAimonen, 2025-03-24:
Ok, found it out. Added this to my `rp2350.cfg` for non-secure debugging:

```
rp2350.dap.core0 configure -event examine-start {
    rp2350.dap apsel 0x2000
    rp2350.dap apcsw 0x43000000
}

rp2350.dap.core1 configure -event examine-start {
    rp2350.dap apsel 0x4000
    rp2350.dap apcsw 0x43000000
}
```

closed 2025-03-24 by PetteriAimonen

lurch, 2025-03-24:
@P33M Is this something that we need to add to the debugprobe documentation?

P33M, 2025-03-24:
This is a property of the chip, so should go in the datasheet or an app note - i.e. RP2350 pays attention to (and enforces) accesses based on HPROTn.

But there's more interactions here than just examining a non-secure core. OTP controls whether or not secure debug is hard-disabled and/or re-enabled by a key - so you would need custom configs for each of those cases. We probably need examples of such.

PetteriAimonen, 2025-03-24:
I have suggested some changes to the example scripts in pull req #131.