raspberrypi/openocd #137: OpenOCD Fails to Connect to RP2040 via CMSIS-DAP Debug Probe

bartola-valves, 2025-07-09:
### 🔧 Setup Summary

- **Target**: Raspberry Pi Pico (brand-new unit, tested multiple)
- **Probe**: Raspberry Pi Debug Probe (CMSIS-DAP firmware v2.0.0)
- **Host OS**: Windows 10/11 (tried on multiple machines)
- **OpenOCD Version**: 0.12.0+dev-gebec950-dirty (bundled via Pico SDK in Visual Studio Code)
- **Invocation Command**:

`openocd -s C:\Users\amogl.pico-sdk\openocd\0.12.0+dev\scripts -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "adapter speed 1000; init; shutdown"
`

The output error of executing the "Flash Project (SWD)" command of the Rasberry Pi Pico extension in Visual Studio Code is: 

[image]

```
 *  Executing task: C:\Users\amogl/.pico-sdk/openocd/0.12.0+dev/openocd.exe -s C:\Users\amogl/.pico-sdk/openocd/0.12.0+dev/scripts -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c adapter speed 5000; program "c:/Users/amogl/OneDrive/Stash Win/Pico/code/Eurorack/NyX/NyX/build/NyX.elf" verify reset exit 

Open On-Chip Debugger 0.12.0+dev-gebec950-dirty (2024-09-27-16:29)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E6647C740388492F
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 5000 kHz
Warn : Too long SWD WAIT, issuing DAPABORT
Warn : Too long SWD WAIT, issuing DAPABORT
Warn : Too long SWD WAIT, issuing DAPABORT
Error: Failed to connect multidrop rp2040.dap0
in procedure 'program'
** OpenOCD init failed **
shutdown command invoked

 *  The terminal process "C:\Users\amogl\.pico-sdk\openocd\0.12.0+dev\openocd.exe '-s', 'C:\Users\amogl/.pico-sdk/openocd/0.12.0+dev/scripts', '-f', 'interface/cmsis-dap.cfg', '-f', 'target/rp2040.cfg', '-c', 'adapter speed 5000; program "c:/Users/amogl/OneDrive/Stash Win/Pico/code/Eurorack/NyX/NyX/build/NyX.elf" verify reset exit'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

I’ve changed every single hardware variable:

- New Pico boards (factory fresh)
- Fresh Debug Probe (new out of box)
- New JST-SH cables - what comes on the Debug Probe.
- Different Windows machines Yet the error remains identical across all setups.

✅ Debugging and Sanity Checks

- Bootsel mode confirmed on target (to avoid firmware interference)
- Target successfully runs blink.uf2 (GPIO confirmed functional)
- Debug Probe firmware is up-to-date (v2.0.0)
- OpenOCD launched manually from CLI with correct script path
- Lowered adapter speed to 100 kHz
- Confirmed matching ground plane and USB power sourcing
- Pico’s RUN pin toggled for reset-to-SWD access

🤔 Hypotheses
- 
- SWD multidrop support introduced in OpenOCD 0.12.0 may interfere with single-target RP2040
- CMSIS-DAP handshake timing mismatch or USB HID driver conflict
- Debug Probe firmware may be incompatible with latest OpenOCD version
- Potential change in VSC Cortex-Debug extension or Pico SDK integration

🙏 Request

Would love insight from maintainers and community on:
- Known issues with rp2040.dap0 connection failures
- Ways to disable SWD multidrop and force single-target handshake
- Recommended OpenOCD version or commit for Debug Probe compatibility
- Any debugging flags that can provide deeper trace of DAP protocol traffic

Thanks in advance!

P33M, 2025-07-09:
The latest version of the debugprobe firmware is not v2.0.0. Please ensure you are using the v2.2.3 release.

bartola-valves, 2025-07-09:
> The latest version of the debugprobe firmware is not v2.0.0. Please ensure you are using the v2.2.3 release.

It's the commercial debugprobe, how do I update the firmware if it doesn't have a bootsel switch?

P33M, 2025-07-09:
https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html#updating-the-firmware-on-the-debug-probe

P33M, 2025-07-09:
That's the CMSIS DAP protocol version and not the probe firmware version. Unfortunate that they can collide. The probe version is reported in the bcdDevice field of the USB device descriptor but no matter.

What happens if you invoke rp2040-rescue.cfg and restart openocd afterwards with rp2040.cfg?

bartola-valves, 2025-07-09:
Well, it seems there is no rescue config on the version bundled in VSC: 

```
PS C:\Users\amogl\OneDrive\Stash Win\Pico\code\Eurorack\NyX\NyX> openocd -s C:\Users\amogl\.pico-sdk\openocd\0.12.0+dev\scripts -f interface/cmsis-dap.cfg -f target/rp2040-rescue.cfg
Open On-Chip Debugger 0.12.0+dev-gebec950-dirty (2024-09-27-16:29)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
embedded:startup.tcl:28: Error: Can't find target/rp2040-rescue.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 28
PS C:\Users\amogl\OneDrive\Stash Win\Pico\code\Eurorack\NyX\NyX> openocd -s C:\Users\amogl\.pico-sdk\openocd\0.12.0+dev\scripts -f interface/cmsis-dap.cfg -f target/rp2040.cfg
Open On-Chip Debugger 0.12.0+dev-gebec950-dirty (2024-09-27-16:29)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Warn : An adapter speed is not selected in the init scripts. OpenOCD will try to run the adapter at very low speed (100 kHz).
Warn : To remove this warnings and achieve reasonable communication speed with the target, set "adapter speed" or "jtag_rclk" in the init scripts.
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E6647C740388492F
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 100 kHz
Warn : Too long SWD WAIT, issuing DAPABORT
Warn : Too long SWD WAIT, issuing DAPABORT
Warn : Too long SWD WAIT, issuing DAPABORT
Error: Failed to connect multidrop rp2040.dap0
```

lurch, 2025-07-10:
> However, I've spotted that the openocd build shipped with the extension is way out of date.

ping @will-v-pi (for when you'e feeling better)

P33M, 2025-07-14:
It isn't, but it's immaterial - the error is the same in both cases. I can replicate the errors with a disconnected SWDIO line. You will also get this behaviour if the pad input is stuck-high or the SWDIO/SWCLK pad output drive is broken. As you have a scope, with SWDIO disconnected can you check:
* swclk
* swdio driven from the debug host
* swdio driven from the debug target

P33M, 2025-07-15:
Are you using the debugprobe_on_pico.uf2 from https://github.com/raspberrypi/debugprobe/releases/tag/debugprobe-v2.2.3 ?
Note that the default swd GPIO assignments on Pico are GPIO2 swclk and GPIO3 swdio and shown pictorially in https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf figure 9.

Even so, the swdio pin that's captured on DIO0 looks a bit glitchy, but without the corresponding clock I can't draw any conclusions.