raspberrypi/openocd #141: QSPI Flash id = 0x154068 not recognised

fischpj, 2025-08-29:
Hi,

I just got a new batch of Pico's from Amazon and I'm having an issue using the official RP Debug Probe with my new Pico's.  I'm on a Mac, Sonoma, running VSCode and the Pico extension.  Compile and run tests via USB work fine, so that verifies that the Pico under test is good.  Debugging, using the probe is a different issue.  My terminal output when I initiate a debug session is below.

I posted an issue in Element14's forum and they suggested that I post here.  Apparently, OpenOCD has the flash chip types embedded in their code and the new Picos are apparently using an unknown flash chip.  **Error: RP2040 rev 2, QSPI Flash id = 0x154068 not recognised**

I think it would be a simple fix to add that new flash's credentials to OpenOCD source and rebuild, before a bunch of new Pico users get cranky.

Can someone please help with the build?

[2025-08-28T23:18:55.821Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
/Users/fischp/.pico-sdk/openocd/0.12.0+dev/openocd.exe -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /Users/fischp/.pico-sdk/openocd/0.12.0+dev/scripts -f /Users/fischp/.vscode/extensions/marus25.cortex-debug-1.12.1/support/openocd-helpers.tcl -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "adapter speed 5000"
Open On-Chip Debugger 0.12.0+dev (2025-07-17-12:44)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'gdb port', not 'gdb_port'
DEPRECATED! use 'tcl port' not 'tcl_port'
DEPRECATED! use 'telnet port', not 'telnet_port'
CDLiveWatchSetup
Info : [rp2040.core0] Hardware thread awareness created
Info : [rp2040.core1] Hardware thread awareness created
adapter speed: 5000 kHz
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E663AC91D37B8939
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 5000 kHz
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x10000001
Info : [rp2040.core0] Cortex-M0+ r0p1 processor detected
Info : [rp2040.core0] target has 4 breakpoints, 2 watchpoints
Info : [rp2040.core0] Examination succeed
Info : [rp2040.core1] Cortex-M0+ r0p1 processor detected
Info : [rp2040.core1] target has 4 breakpoints, 2 watchpoints
Info : [rp2040.core1] Examination succeed
Info : [rp2040.core0] starting gdb server on 50000
Info : Listening on port 50000 for gdb connections
Info : accepting 'gdb' connection on tcp/50000
Info : no SDFP found
**Error: RP2040 rev 2, QSPI Flash id = 0x154068 not recognised**
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
shutdown command invoked
[2025-08-28T23:18:56.698Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...

P33M, 2025-08-29:
Are these Picos official Raspberry Pi products?

fischpj, 2025-08-29:
Yes they are, purchased from Amazon about a week ago.  (c) 2020 on the silk screened on the board.

P33M, 2025-08-29:
Can you post a picture of the underside of the board, in particular with a closeup of the laser-etched data matrix under the FCC logo?

fischpj, 2025-08-29:
Here's the back

[image]

fischpj, 2025-08-29:
Wow!  Ok, I'll call Amazon.  Thank you for your quick reply

fischpj, 2025-08-29:
I was looking at the Raspberry Pi Pico manual PDF and their pictures match mine, as do the pictures on Newark's product page.  None I've found look like yours.  I'm just confused and want to buy real picos.

[image]

P33M, 2025-09-01:
Regarding your Picos sourced from Amazon, I would recommend returning these for a refund. In your support case, specifically refer to these being counterfeits. Separately, we've raised this with Amazon as a business seller.